WellDone/MoMo-Firmware #171: LDF file / firmware version mismatch causes system log functionality to be useless

amcgee, 2015-02-01:
We should probably log the error and fall back to just printing the raw value without doing type-specific parsing, that way we at least get the rest of the log and know where this occurs.

```shell
$ momo controller read_log add_ldf ./momo_modules/mainboard/src/log_definitions.ldf add_ldf ./momo_modules/shared/pic24/src/log_definitions.ldf display
ArgumentError: Parameter has unknown type
Additional Information:
type: fw_scheduled_task
param: Scheduled Call
```

timburke, 2015-02-01:
This is because the log needs to know about types defined in the python 
package alongside the shared/pic24 module. It shouldn't have anything to 
do with either read_log or display.  It's just telling you that your ldf 
file refers to python types that aren't defined.

You shouldn't add an ldf file without also adding its corresponding type 
library.
Corrected command would be:

momo import_types ./momo_modules/shared/pic24/python/firmware_types controller read_log add_ldf ./momo_modules/mainboard/src/log_definitions.ldf add_ldf ./momo_modules/shared/pic24/src/log_definitions.ldf display

You can also do this automatically using the momo rcfile:
> timburke$ momo --rcdir
> /Users/timburke/Library/Preferences/WellDone-MoMo/momo_config.txt
> $ cat /Users/timburke/Library/Preferences/WellDone-MoMo/momo_config.txt
> [root]
> import_types 
> /Users/timburke/Projects/MoMo-Firmware-Personal/momo_modules/shared/pic24/python/firmware_types
>
> [MIBController.SystemLog]
> add_ldf 
> /Users/timburke/Projects/MoMo-Firmware-Personal/momo_modules/mainboard/src/log_definitions.ldf
> add_ldf 
> /Users/timburke/Projects/MoMo-Firmware-Personal/momo_modules/shared/pic24/src/log_definitions.ldf

amcgee, 2015-02-01:
Ok, but it still shouldn't be a fatal error.